equinor/isar #73: Upload inspections results in collect instead of cancel

vetlek, 2021-11-05:
Currently all the inspections results are uploaded on the end of the mission when we enter cancel. This can be problematic if the mission is very long and we want the results as fast as possible (no need to get all inspections at the same time). If the state machine crashes due to internal or external issues, the inspection results are potentially lost. A potential fix is to upload the results in the collect state.

aestene, 2021-11-08:
Note that there are requirements for a continuous internet connection by making this change. This is to some extent already required, so it may not be much of a concern. 

The upload now is also dependent on the full format of the mission. Meaning it is easier to know all the results when uploading than to do one-by-one. It is not impossible to implement however.

aestene, 2021-12-14:
Due to the nature of the final result which should be updated and the fact that the upload is not valid before the final package is put together it may be that the current solution is preferred. However, we may perform download in collect and try to use a local file cache for storing the files until they should be uploaded. 

In case of a mission failing all the collected results may be uploaded at the end.

PR #189 (2022-01-31): Upload in separate thread to multiple storage handlers
Storage may now be configured as a list specifying modules that should be used for storage. The state machine will send any collected inspection to a separate uploader thread which handles uploads. It will loop through all selected storage handlers and upload the inspection to each one.

In addition a new storage handler has been created to support the SLIMM ingestion API.

Closes #73
closed 2022-01-31 by aestene