equinor/isar #131: Failing a task should invalidate related subsequent tasks

aestene, 2021-12-02:
**Describe the improvement you would like to see**
If the robot fails to execute a task any subsequent tasks that are dependent on the failed task should also fail. 

**How will this change existing functionality?**
When a task fails as part of a mission ISAR currently moves on to the next task and executes that. So if the robot fails to execute a DriveTo task it will then perform the next TakeImage task as if the previous DriveTo succeeded. This leads to incorrect data being gathered.

**How will this improvement affect the current Threat Model?**
The functionality should not have any immediate effect on the current threat model.

aestene, 2021-12-02:
Can we introduce a concept where we do something like the following:

Introduce a `depends_on` variable for all steps which contains a list of other step instances that it is dependent on. 
-  ISAR would then verify that all steps in the `depends_on` list has been completed before allowing the step to be sent. 
- If all dependent steps are not complete the step will be marked as failed.

aestene, 2021-12-02:
Another alternative could be to add another hierarchy level to missions such that certain tasks contain several steps.
So a mission contains a list of tasks and each task contains a list of steps that depend on eachother. If one step in a task fails all steps in the task fail and the task automatically also fails.

knord94, 2021-12-07:
A third alternative, similar to the _task_ dependency, is a _criteria_ dependency, which will be robot specific and handled at robot level instead of ISAR. For instance, `TakeImage` could include a position (the robot/camera should be located for taking the image) which would be a criteria.

closed 2021-12-09 by vetlek