equinor/isar #135: Retrieve configuration for specific sites from outside repository

aestene, 2021-12-02:
**Describe the new feature you would like to see**
Site specific configuration like map, metadata, predefined_poses etc. should be retrieved from storage and not be part of the version control in the git repository.

**Describe the solution you'd like**
A schema determines how to define the different data types (map for example) which is then written in a JSON file. The JSON file is collected from a remote location by specifying path/location/similar.

**How will this feature affect the current Threat Model?**
Should not affect the current threat model, but care should be taken with authentication if a remote location is used for storing the configuration.

aestene, 2022-07-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

MortFred, 2022-08-22:
Outdated

closed 2022-08-22 by MortFred