equinor/isar #152: Rename send state

aestene, 2021-12-07:
**Describe the improvement you would like to see**
The `Send` state does not reflect the actions of the state very well as there is confusion as to what is sent where. Find a better name which properly reflects the actions of the state.

**How will this change existing functionality?**
No change other that naming convention.

**How will this improvement affect the current Threat Model?**
No change.

aestene, 2022-01-03:
Suggestions
- `SendTask`
- `DispathTask`
- `ScheduleTask`

aestene, 2022-02-22:
This issue has automatically been marked as stale as there has been no activity for 60 days. Remove stale label or comment or this will be closed in 7 days.

nicholasdalhaug, 2022-03-07:
Still something we need to solve.

closed 2022-04-06 by anetteu