equinor/isar #157: Consider using 'type' instead of 'name' for type of task in a mission

tsundvoll, 2021-12-08:
**Describe the improvement you would like to see**
Today we tag each task in a mission with a 'name'. Probably a more precise wording would be 'type'.
For example for DriveToPose. The resulting predefined mission would look something like this.
````
"tasks": [
        {
            "type": "drive_to_pose",
            "pose": {
                "position": {
                    "x": -3.6,
                    ....
            }
        },
````

**How will this change existing functionality?**
Nothing, just a refactor

**How will this improvement affect the current Threat Model?**
Nothing

PracticalMetal, 2021-12-08:
Hey, I would like to work on this, can you suggest me ways to get started?

tsundvoll, 2021-12-09:
> Hey, I would like to work on this, can you suggest me ways to get started?

That is great @PracticalMetal. You can start by having a look at
`src/robot_interface/models/mission/task.py`.
Here we define different dataclasses that are different types of tasks that can make up a robot mission. This issue is to rename the `name` attribute of those dataclasses to `type` instead.

Furthermore, we have some example missions in the
`src/isar/config/predefined_missions/` folder.
The tasks listed in those files also need to swap `name` with `task`.

That should be it for this issue. We are happy to receive a PR on this from your forked version of this repo.

PracticalMetal, 2021-12-11:
Sorry for the late reply, I will be working on this. Thanks!

PR #239 (2022-04-08): Rename `task.name`to `task.type`
The previous `task.name` is not a unique name for a task, but a
description of the type of task, for example `take_image`.

Resolves #157
closed 2022-04-08 by anetteu