equinor/isar #182: Change default data configuration to internal

aestene, 2022-01-05:
**Describe the improvement you would like to see**
The default value for data configuration should be set to `internal` instead of the current value `open`.
https://github.com/equinor/isar/blob/ff08e1d7d955e051ee4bf1da82ad9cae05531156/src/isar/config/default.ini#L143

**How will this change existing functionality?**
No change, just correct metadata.

**How will this improvement affect the current Threat Model?**
The provided results to storage will be treated as internal by default which is the correct behavior.

pramod74, 2022-01-13:
Hi @aestene,

I would like to work on this issue. Kindly assign it to me.

Thanks,
Pramod

aestene, 2022-01-14:
Great @pramod74! We appreciate your interest in the project!

pramod74, 2022-01-14:
Thanks, @aestene.

I have submitted a PR with the suggested fix. Kindly review.

Thanks,
Pramod

aestene, 2022-02-17:
Closed by #185