equinor/isar #244: Can't run isar as container with non-default robot package

GodVenn, 2022-04-19:
**Describe the bug**
When changing the robot type in the docker-compose file, we get the error ` ModuleNotFoundError: No module named 'isar_robot'`.
Somewhere in the code, we are missing a check for environment variables so that it defaults to isar_robot anyways.

**To Reproduce**
Steps to reproduce the behavior:
Change `ROBOT_REPOSITORY_CLONE_URL`  from "isar-robot" to "isar-turtlebot" and `ROBOT_PACKAGE` from "isar_robot" to "isar_turtlebot".

**Expected behavior**
ISAR should run in a container using the isar turtlebot library.

aestene, 2022-04-19:
There was a change after the docker-compose file was made that introduced a different way of handling environment variables and settings. It now requires that environment variables be prefixed by "ISAR" such that it should be `ISAR_ROBOT_PACKAGE` as the environment variable in docker-compose. @GodVenn

aestene, 2022-04-19:
Have a look at this file and line for reference: https://github.com/equinor/isar/blob/main/src/isar/config/settings.py#L206

PR #245 (2022-04-20): Add 'ISAR' prefix to compose file
Closes #244
closed 2022-04-20 by GodVenn