equinor/isar #270: Add return to home / docking functionality

aestene, 2022-05-20:
**Describe the new feature you would like to see**
There should be an endpoint for a robot to return to its home location. Home location could be docking or just a specified position.

**Describe the solution you'd like**
- Consider if the position should be specified in the endpoint or be a configured variable
- Run a mission on the robot with a `DockingProcedure`

**How will this feature affect the current Threat Model?**
N/A

vetlek, 2022-05-25:
The feature that is described in the issue seems like it is covered by the `drive_to_pose` endpoint (have to specify the location). Should the user be able to specify that the robot should return to home at the end of the mission (add a task on the end that returns the robot to home + possible docking)?

vetlek, 2022-05-31:
A suggested solution for this issue can be found in the linked PR. Not sure if this is a satisfactory solution, but it gives an optional way to specify a pose that the robot should return to at the end of the mission. If we want tot return to home outside of a mission, the drive_to endpoint can be used.

PR #281 (2022-06-09): Add return to specific pose at end of mission
Possibly Closes #270
closed 2022-06-09 by vetlek