equinor/isar #275: Consider adding limit on retries on stopping steps

vetlek, 2022-05-27:
**Describe the improvement you would like to see**
Currently there are no upper limit on the number of retries that are done when we are trying to stop a certain step. This will make the robot go in an infinite loop until we are successful, while an upper limit makes it possible that we leave the state without actually having stopped the robot. The desired behavior when stopping/pausing should be evaluated and an upper limit should be implemented if it is found desirable.

**How will this change existing functionality?**
Possibly add a maximum number of retries.

**How will this improvement affect the current Threat Model?**
No change.

aestene, 2022-07-27:
This issue has automatically been marked as stale as there has been no activity for 60 days.

GodVenn, 2022-08-08:
Still desired funtionality - removing stale flag

eivindsjovold, 2022-08-22:
Closed with PR #308. Follow-up issue on how to handle this in #313 .

closed 2022-08-22 by eivindsjovold