equinor/isar #277: Run get_inspections in separate thread

vetlek, 2022-05-27:
**Describe the improvement you would like to see**
Currently `get_inspections` are run when the robot successfully finish a step in the monitor state. This is a function that can potentially block for a long time and it is unnecessary that it runs in the same thread as the state machine since this can be done while the robot is doing other operations. This functionality should be moved to a separate thread that the state machine communicates with by sending a copy of the steep in queue.  

**How will this change existing functionality?**
Make the state machine more responsive and efficient.

**How will this improvement affect the current Threat Model?**
No change

aestene, 2022-07-27:
This issue has automatically been marked as stale as there has been no activity for 60 days.

GodVenn, 2022-08-08:
Removing stale label because we still want this

closed 2023-02-01 by monkeyysups