equinor/isar #321: Version requirements in `setup.py`

luteberget, 2022-09-20:
**Describe the bug**
`setup.py` is missing versions requirements for external packages. This causes issues if you have old versions already installed.

**To Reproduce**
Install `alitra==1.0.16` and run ISAR:

```
~> isar
Traceback (most recent call last):
  File "/home/bjornalu/repos/isar/main.py", line 9, in <module>
    from isar.apis.api import API
  File "/home/bjornalu/repos/isar/src/isar/apis/api.py", line 14, in <module>
    from isar.apis.models.models import StartMissionResponse
  File "/home/bjornalu/repos/isar/src/isar/apis/models/__init__.py", line 1, in <module>
    from .models import InputPose, StartMissionResponse
  File "/home/bjornalu/repos/isar/src/isar/apis/models/models.py", line 4, in <module>
    from alitra import Frame, Orientation, Pose, Position
ImportError: cannot import name 'Frame' from 'alitra' (/home/bjornalu/.local/lib/python3.9/site-packages/alitra/__init__.py)
```

Suggest to modify `setup.py` like this

```
    install_requires=[
        "alitra>=1.1",
```

... and probably other packages should have version numbers too. Every few months I get version issues and need to guess which packages must be upgraded. Am I doing it wrong?

aestene, 2022-09-20:
You are right that this should be added. I have added it for `alitra` for now and we will consider adding it for the others where we see this is necessary. 

Thanks for the feedback!

aestene, 2022-09-20:
Closed by #325

closed 2022-09-20 by aestene