equinor/isar #322: ISAR MQTT client crashes

luteberget, 2022-09-20:
**Describe the bug**
Starting ISAR, I get this error message, and the MQTT client does not connect.
```
Exception in thread ISAR MQTT Client:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/bjornalu/repos/isar/src/isar/services/service_connections/mqtt/mqtt_client.py", line 53, in run
    self.connect(host=self.host, port=self.port)
  File "/home/bjornalu/.local/lib/python3.9/site-packages/backoff/_sync.py", line 129, in retry
    _call_handlers(on_success, **details)
  File "/home/bjornalu/.local/lib/python3.9/site-packages/backoff/_sync.py", line 21, in _call_handlers
    hdlr(details)
TypeError: 'staticmethod' object is not callable
`` `

**To Reproduce**
Start Isar.

**Expected behavior**
The problem goes away if I add `.__func__` to the static method references in `src/isar/services/service_connections/mqtt/mqtt_client.py` line 107:

```
    @backoff.on_exception(
        backoff.expo,
        ConnectionRefusedError,
        max_time=300,
        on_success=_on_success.__func__,
        on_backoff=_on_backoff.__func__,
        on_giveup=_on_giveup.__func__,
    )
    def connect(self, host: str, port: int) -> None:
```

aestene, 2022-09-21:
I'm not able to recreate this issue on my own system. I'm using python 3.10 however while I see you're on 3.9. Unless there is some reason you have to use 3.9 I would recommend trying with 3.10. If there is an issue there we should also set the requirement for ISAR to 3.10.

luteberget, 2022-09-21:
I think you are right that it's about python version. The following works on python 3.10:

```python
class X:
  @staticmethod
  def x():
    pass
  y = [x]

(X.y[0])()
```

... but on python 3.9, fails with
```
TypeError: 'staticmethod' object is not callable
```

PR #356 (2022-10-25): Revert from staticmethods
to support Python 3.9 functionality. The static methods are not really needed here.

Resolves #322 for Python 3.9.
closed 2022-10-25 by tsundvoll