equinor/isar #324: Robot state MQTT messages not retained?

luteberget, 2022-09-20:
**Describe the improvement you would like to see**
If I start my own program that interfaces with ISAR *after* ISAR has started, then the robot state might be idle, and then I never get a new `state` message over MQTT. Previously, these messages were "retained", so that you would get them again when you re-connect. For example, when I restart my program, I would like to know the current state.

I see that you made the state messages not-retained. Why is that? Is there some other way I should be reading the state?

**How will this change existing functionality?**
Don't know.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

luteberget, 2022-09-20:
The change I am referring to is 71741f7caf7925e6f8b023004606b3d9366714d7

aestene, 2022-09-21:
The retain change was primarily focused on the step, task and mission topics and we should revert the change for the state topic so retain can be enabled for that one.

The reason for not using retain with the others is that you would receive the status of a potentially old mission.

PR #327 (2022-09-21): Add option to disable tls through MQTT
Also sets retain to true for the state topic.
Closes #324 and closes #323
closed 2022-09-21 by aestene