equinor/isar #328: Why does stop_mission require not being in the initializing state?

luteberget, 2022-09-22:
**Describe the improvement you would like to see**

To use the HTTP API correctly, it seems that you need to know the state (`Idle`, `Monitor`, etc.). Some API calls are only allowed/available when in some specifc states. From how I read the code (https://github.com/equinor/isar/blob/4699f55f6f012da44bf30bd56398d923d943984b/src/isar/apis/schedule/scheduling_controller.py) understand the constraints are:

 * `start_mission_by_id`: must be in state `Off` or `Idle`.
 * `start_mission`: must be in state `Off` or `Idle`.
 * `pause_mission`: must be in state `InitiateStep`, `Monitor` or `Off`.
 * `resume_mission`: must be in state `Off` or `Paused`.
 * `stop_mission`: must be in state `Off`, `InitiateStep`, `Monitor`, `Paused`, `StopStep`.
 * `drive_to`: must be in state `Off` or `Idle`.

Unless this is already available somehow, I suggest to add something like this list to the static docs or the interactive API docs. 

Some questions:

 * Will the API calls work in the `Off` state? If not, is there a correct HTTP response for this?
 * Why not `pause_mission` in `StopStep` or `Initialize`?
 * Why not `stop_mission` in `Initialize`?

Christdej, 2022-10-14:
Good points you bring up here, I have added the issue to our backlog and board.

aestene, 2023-01-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Christdej, 2023-01-04:
We will for now not add information about what state you need to be in to call an API as you get feedback from the API if you are in an incorrect state. Small changes has been done in #378

PR #378 (2023-01-05): State machine diagram and small state machine fixes
Closes #328
closed 2023-01-05 by Christdej