equinor/isar #341: Publish isar status over mqtt

oysand, 2022-10-13:
**Describe the new feature you would like to see**
ISAR should periodically update its status over mqtt.

**Describe the solution you'd like**
If you have an idea of how to implement the feature please describe it here.

**How will this feature affect the current Threat Model?**
N/A

aestene, 2022-10-19:
This issue is very related to https://github.com/equinor/flotilla/issues/426 and requires implementation both in ISAR and Flotilla.

Christdej, 2022-11-02:
Could it be an idea to publish the whole mission each time something changes, and rather update the status of completed tasks and steps? Fixes problem where if we miss one update from ISAR, the status in flotilla does not match status in ISAR.

EinarUeland, 2022-11-02:
@oysand: 
We already have mission_status (which I assume serve a different purpose)

Could you specify what status that you want ISAR to publish over MQTT?
Robot online & No connection to robot? or a more detailed status?

aestene, 2022-11-03:
@EinarUeland I think there are two major things the status message should accomplish and it is 
1. Let Flotilla know that it is able to reach the ISAR instance (so the ISAR instance is running)
2. Let Flotilla know that ISAR is able to reach the robot (this requires a function in the robot interface with a "ping" functionality I think)

So in essence I think the status message should accomplish those things.

Other things to include
- Robot name
- Robot capabilities maybe?
- ISAR host address and port?
- State machine current state?

aestene, 2022-11-03:
- Currently executing mission ID (or None if there is no mission in progress)

closed 2023-02-08 by aestene