equinor/isar #351: Empty response body on OK request in scheduling controller

eivindsjovold, 2022-10-19:
**Describe the bug**
Pause-, resume- and stop mission endpoints in scheduling controller returns "null" in response body for response 200. Mission is successfully pause, stopped or resumed, but the backend throws an error because of the empty response body.

**To Reproduce**
Schedule a mission in frontend -> pause, resume or stop mission and see error messages.

**Expected behavior**
Scheduling api should not return an empty response body on successful stop, pause or resume.

**Screenshots**
[image]

GodVenn, 2022-10-19:
[image]
This code should be changed so that an empty string is accepted if that is the case

GodVenn, 2022-10-19:
[image]
This code has been copy pasted from the "Start-Mission" endpoint to the others.
That is the reason for the bug; the other endpoints should not expect to get a mission response from ISAR.

PR #353 (2022-10-20): Add responses to pause, resume, stop mission endpoints
Closes #351.
closed 2022-10-20 by eivindsjovold