equinor/isar #368: Remove

tsundvoll, 2022-11-24:
**Describe the improvement you would like to see**
A clear and concise description of the improvement proposal.

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2023-01-24:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Christdej, 2023-02-15:
Would it make more sense to change TimeIndexedPose to Pose and keep "start_time"?

tsundvoll, 2025-01-13:
> Would it make more sense to change TimeIndexedPose to Pose and keep "start_time"?

This is implemented. Closing the issue

closed 2025-01-13 by tsundvoll