equinor/isar #374: docker-compose-turtlebot.yml out of date

tsundvoll, 2022-12-02:
**Describe the bug**
With the new way we build the robot packages (the robot package builds on top of the isar base), this docker-compose is outdated. It can for example be updated to build from the isar-turtlebot package instead (if we still want this compose file here).

**To Reproduce**
'docker-compose -f docker-compose-turtlebot.yml up --build '

**Expected behavior**
Isar starts with the isar-turtlebot installed besides it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

aestene, 2023-01-11:
I would suggest we remove this file from ISAR.

aestene, 2023-06-20:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2025-03-05:
The docker-compose-turtlebot.yml file is removed in https://github.com/equinor/isar/pull/749.

To run with turtlebot, see the isar-turtlebot repository