equinor/isar #381: Authentication fails as 0.0.0.0 redirection is not configured for application

anetteu, 2023-01-06:
**Describe the bug**
The redirection URL `http://0.0.0.0:3000/oauth2-redirect` is not configured for the application `IsarOpenAPIAuth`. It is not possible to register `http://0.0.0.0:3000/oauth2-redirect` as a valid redirect for `http`requires `localhost`.

**To Reproduce**
Steps to reproduce the behavior:
Set `ISAR_AUTHENTICATION_ENABLED = true`, start ISAR locally, and authorize.

**Expected behavior**
Authentication should work

**Screenshots**
[image]

anetteu, 2023-01-06:
@oysand Any comments?

GodVenn, 2023-02-13:
This is a matter of adding the URI to our app registration's list of accepted URIs.
I will do this now

GodVenn, 2023-02-13:
Can't add specific IP addresses to that list.
Use localhost instead of 0.0.0.0?

GodVenn, 2023-02-13:
When ISAR is hosted at 0.0.0.0, it is accesible through localhost. By accessing it through localhost, you avoid this issue.
Currently we can't change the uvicorn host to localhost although this was suggested in #403 , but caused issues in docker.

Suggesting to close this issue? @anetteu

oysand, 2023-02-13:
In order to redirect to 0.0.0.0 the page need to be HTTPS, is this something we want to setup?

aestene, 2023-06-20:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Eddasol, 2023-06-26:
Need to verify if this still is an issue. If so, should consider making new issue to implement HTTPS

closed 2023-08-15 by anetteu