equinor/isar #387: Fastapi 0.89.1 fails

tsundvoll, 2023-01-13:
**Describe the bug**
```
venv-robot/lib/python3.10/site-packages/fastapi/routing.py:255: in app
    content = await serialize_response(
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

    async def serialize_response(
        *,
        field: Optional[ModelField] = None,
        response_content: Any,
        include: Optional[Union[SetIntStr, DictIntStrAny]] = None,
        exclude: Optional[Union[SetIntStr, DictIntStrAny]] = None,
        by_alias: bool = True,
        exclude_unset: bool = False,
        exclude_defaults: bool = False,
        exclude_none: bool = False,
        is_coroutine: bool = True,
    ) -> Any:
        if field:
            errors = []
            response_content = _prepare_response_content(
                response_content,
                exclude_unset=exclude_unset,
                exclude_defaults=exclude_defaults,
                exclude_none=exclude_none,
            )
            if is_coroutine:
                value, errors_ = field.validate(response_content, {}, loc=("response",))
            else:
                value, errors_ = await run_in_threadpool(
                    field.validate, response_content, {}, loc=("response",)
                )
            if isinstance(errors_, ErrorWrapper):
                errors.append(errors_)
            elif isinstance(errors_, list):
                errors.extend(errors_)
            if errors:
>               raise ValidationError(errors, field.type_)
E               pydantic.error_wrappers.ValidationError: <exception str() failed>
```

**To Reproduce**
Run `pytest` with fastapi==0.89.1

**Expected behavior**
All tests pass. This works with fastpi==0.88.0

**Screenshots**
N/A

tsundvoll, 2023-01-16:
- [ ] Let each endpoint return just one type, and raise the exception (instead of returning error message)

Possible refactorings:
- [ ] Have similar responses in the Router setup, instead of in each route

oysand, 2023-01-25:
Is this fixed in #388 ?

tsundvoll, 2023-01-25:
Yes, solved by #388

closed 2023-01-25 by tsundvoll