equinor/isar #393: Add api tests that goes deeper (more like an integration test)

tsundvoll, 2023-01-18:
**Describe the improvement you would like to see**
The current way we test the api, e.g. in `test_scheduler_router.py` mocks a lot of the behavior. Find a way to test as deep as possible, including getting the state machine in action and get a response back.

```python
@mock.patch.object(SchedulingUtilities, "get_state", mock_return_idle)
@mock.patch.object(SchedulingUtilities, "get_mission", mock_get_mission)
@mock.patch.object(SchedulingUtilities, "start_mission", mock_void)
def test_start_mission_by_id(self, client: TestClient):
    response = client.post(url=f"{self.schedule_start_mission_path}/1")
    assert response.status_code == HTTPStatus.OK
    assert response.json() == jsonable_encoder(mock_start_mission_response)
```

**How will this change existing functionality?**
Today, the api endpoints puts a command on a queue and returns OK back if that succeeds, more or less. I would like a test that also check correct behavior in the state machine, given an API call

**How will this improvement affect the current Threat Model?**
N/A

tsundvoll, 2023-01-23:
@Christdej might have input and experience on this

aestene, 2023-06-22:
Would be relevent to include the MQTT messages into this testing. Preferrably without using a broker but mocking and checking the messages received.

aestene, 2023-09-05:
Closing this due to age and the fact that we have extended our tests for ISAR in a direction which covers a lot of testing.

closed 2023-09-05 by aestene