equinor/isar #394: Wait for state machine response before returning the HTTP requests in the API

tsundvoll, 2023-01-23:
**Describe the improvement you would like to see**
Today: The API returns immediately after putting commands on the "queue" between the API and the State Machine. For example when trying to schedule a mission, the "start-mission" endpoint might return 200, even before the state machine has managed to start the mission.

I would like the API to wait for the wanted response in the state machine before returning the request.

For example at the end of `start_mission()` in `isar/src/isar/apis/schedule/scheduling_controller`
```python
self.scheduling_utilities.start_mission(
    mission=mission, initial_pose=initial_pose_alitra
)
return mission.api_response()
```

**How will this change existing functionality?**
N/A

**How will this improvement affect the current Threat Model?**
N/A

Christdej, 2023-06-09:
At the moment you are able to queue a mission even if the robot is unavailable. Then the mission will fail instantly. I believe this is fine for now and we dont have to prioritize this at the moment.

aestene, 2023-09-05:
This will be closed for now due to low activity and uncertainty around the need for this functionality.

closed 2023-09-05 by aestene