equinor/isar #434: Enable localization of robot based on user innput

aestene, 2023-03-20:
**Describe the new feature you would like to see**
A robot may require to be localized prior to running the first mission. To make this procedure as user friendly as possible for end users there should be functionality to enable localization without entering robot specific software. 

How this should be solved is still not certain but what is required is:
- The robot should be placed on a specific location on a specific deck of the asset
- The operator should then through the user interface select which deck the robot is on for the currently selected asset and press OK/localize etc.
- The operator should get feedback to tell if the robot is successfully localized
- Optional: The robot should not be displayed as available until the robot has been localized

**Describe the solution you'd like**
Open for interpretation but needs to be as simple to use for the operator as possible.

**How will this feature affect the current Threat Model?**
N/A

aestene, 2023-03-20:
This is related to 
https://github.com/equinor/flotilla/issues/534
https://github.com/equinor/flotilla/issues/535

Eddasol, 2023-05-04:
Outstanding part is sufficiently covered by isar-taurob issue 139

closed 2023-05-04 by Eddasol