equinor/isar #465: Authentication fails on on-behalf-of-flow between flotilla and ISAR. - not sure if flotilla or isar issue

eivindsjovold, 2023-05-21:
**Describe the bug**
New releases after 21.05.2023 has changes to how tokens are read it seems. Since we dont have locked package versions, i suspect we have a package(potentially FastAPI?) who has updated with a breaking change.

Error msg: "Flotilla failed to authorize towards ISAR: {"detail":"Unable to process token"}" 

**To Reproduce**
Steps to reproduce the behavior:
Build a new release on any of the isar-robot packages with authentication flow set up from flotilla and witness the error message:
Flotilla failed to authorize towards ISAR: {"detail":"Unable to process token"}

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

aestene, 2023-07-21:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2025-01-13:
Fixed (a long time ago)