equinor/isar #472: get_inspections should return a List

Afonso-2403, 2023-06-03:
**Describe the improvement you would like to see**
The function [get_inspections](https://github.com/equinor/isar/blob/a8d0c12625d04852735755a60a53f2630be6d277/src/robot_interface/robot_interface.py#LL130C5-L130C5) should return a List of inspections instead of a Sequence.

**How will this change existing functionality?**
This implies that implementations of this function in [isar-taurob](https://github.com/equinor/isar-taurob/blob/202ac8ca1d78f40d94c32a920a5f55bc01706e1e/src/isar_taurob/robotinterface.py#LL106C1-L106C1), [isar-exr](https://github.com/equinor/isar-exr/blob/5902786d86fcb73b548a7b00f0ee6c47036b7be2/src/isar_exr/robotinterface.py#L163), [isar-anymal](https://github.com/equinor/isar-anymal/blob/d089bd7bd8c9597c6a11e4041bbddfc1ea286ecd/src/isar_anymal/robotinterface.py#L64), [isar-turtlebot](https://github.com/equinor/isar-turtlebot/blob/65846214e9c6ec5f29ec760166d59b7857e06c89/src/isar_turtlebot/robotinterface.py#L55) and [isar-robot](https://github.com/equinor/isar-robot/blob/192ee57b36403684619c7b108395f05ff35c9d33/src/isar_robot/robotinterface.py#L94) should also return a List

Eddasol, 2023-06-05:
Why should we do this change?

aestene, 2023-06-23:
The `Sequence` object was used without any specific reason for doing so.
- We are using `List` everywhere else
- We have some operations required on this that requires a `List`
- Note that a `List` is a `Sequence` but a `Sequence` is not a `List`

aestene, 2023-06-23:
Note that this is primarily a type hinting exercise which should be performed in all packages.

closed 2024-01-15 by aestene