equinor/isar #478: Specify file-format in mission sent from flotilla

Christdej, 2023-06-16:
Some inspections give us several different file-formats based on what we want, for instance, an image can be jpeg and png. Today the only way to change this is through changing a config-variable, which requires a restart of the application. It would be beneficial to be able to send missions with different file-formats without restarting the application.

Options:
Add a field to a step called i.e. `file_format` which would have to be added to a `StartMissionDefinition` and propagated through isar into a step.

Use the metadata field already present to add i.e. `{"sequence_thermal_video": true}` which again can be caught by the robots.

Labeled as "breaking-change" as it might be breaking dependent on solution chosen

aestene, 2023-06-26:
I'm not sure I agree with this solution. 

From the planning part of the mission it should not be very relevant which file format the result should be given in. This is very robot specific and will vary from robot to robot how it would function. If anything, we would like to change the file format based on which type of analysis should be run on the robot but that can be propagated (and should already be done so through https://github.com/equinor/flotilla/issues/868) such that isar-robot knows which analysis should be run and can use that information if required. 

We have one case where this matters now and given the path we see going forward it might not be relevant anymore either. I would consider closing the issue until then.

aestene, 2023-09-05:
Closing unless there is further comments @Christdej ?

closed 2023-09-05 by aestene