equinor/isar #481: Bump pydantic to version 2.x

eivindsjovold, 2023-07-12:
**Describe the improvement you would like to see**
Pydantic is currently frozen to the latest version 1. Version 2 har some breaking changes, but we would like to bump it still.

**How will this change existing functionality?**
Some import structures are changed in pydantic v2.

**How will this improvement affect the current Threat Model?**
In general, latest version of dependencies improve the overall threat model by reducing the number of vulnerabilities.

aestene, 2023-09-10:
This issue has automatically been marked as stale as there has been no activity for 60 days.

oysand, 2023-10-02:
Consider unlocking pydantic version at the same time

PR #490 (2023-10-12): Fix Key Error on analysis type and unlock packages
Closes #481 

Changes done according to:
https://docs.pydantic.dev/dev-v2/migration/#validator-and-root_validator-are-deprecated
closed 2023-10-12 by Christdej