equinor/isar #510: Migrate from OpenCensus to OpenTelemetry

tsundvoll, 2024-01-09:
**Describe the improvement you would like to see**
Migrate from OpenCensus to OpenTelemetry, as the former is deprecated.

Reference and migration guide: https://learn.microsoft.com/en-us/azure/azure-monitor/app/opentelemetry-python-opencensus-migrate?tabs=aspnetcore

**How will this change existing functionality?**
The functionality should be the same; i.e. telemetry and logs should still be sent to Application Insight.

**How will this improvement affect the current Threat Model?**
Will possibly improve the Threat Model, since the opencensus package seem to introduce a dependency on gcc that has some vulnerabilities.

"After July 31st, 2023, the OpenCensus project will no longer be maintained. This means that there will be no new features added to the project, and any security vulnerabilities that are found will not be patched." - https://opentelemetry.io/blog/2023/sunsetting-opencensus/

tsundvoll, 2024-01-12:
It seems like this SDK for Python is not quite mature yet. After some investigations and trials, I experienced it being slow to start the application, and sometimes not starting at all. When I Ctrl+C it looked like it was interrupted at some `if lock.acquire(block, timeout)` in a thread. Could be related to our threads, or something going on within the `configure_azure_monitor()` function (with and without arguments) which I tried to run at startup.

I also got `Exception  in detector <opentelemetry.resource.detector.azure.vm.AzureVMResourceDetector object at 0x1318b1e20>, ignoring` and I could not find good explanations of this in the documentations / forums.

According to Microsoft (https://learn.microsoft.com/en-us/azure/azure-monitor/app/opentelemetry-python-opencensus-migrate?tabs=aspnetcore) they support the old OpenSensus SDK until September 30 2024.

I therefore suggest we postpone the migration to OpenTelemetry until the Python SDK is more mature and stable.

aestene, 2024-05-03:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2025-01-27:
Investigate if this SDK is ready now

PR #820 (2025-07-29): Migrate to OpenTelemetry
## Ready for review checklist:
- [x] A self-review has been performed
- [x] All commits run individually
- [x] Temporary changes have been removed, like logging, TODO, etc.
- [x] The PR has been tested locally
- [ ] A test has been written
  - [x] This change doesn't need a new test
- [x] Relevant issues are linked
- [ ] Remaining work is documented in issues
  - [ ] There is no remaining work from this PR that requires new issues
- [x] The changes do not introduce dead code as unused imports, functions etc.

Closes #510