equinor/isar #525: Publish descriptive MQTT message on upload to Blob Storage

tsundvoll, 2024-02-28:
**Describe the new feature you would like to see**
When an inspection result is stored in Blob Storage, a reference to the data should be provided in a message on the "inspection_result" topic over MQTT. It should be possible for another system to later retrieve the inspection result from the blob storage using only the information provided in the message.

**Describe the solution you'd like**
When an image or another inspection result is stored in a blob storage, retrieve information about how this can be accessed later. Part of the issue is to figure out technically how this is done with blob storage, whether it is an ID or an url etc.

Make sure this ID / URL / other reference is published over MQTT on the "inspection_result" topic. Something similar is already in place, but I suspect the ID provided is not working.

A good starting point is the `isar/storage/uploader.py` and `isar/storage/blob_storage` modules, where a message is published to the "inspection_result" topic.

Verify that such a message can be received and that the information provided is enough to retrieve the inspection result back from the blob storage.

**How will this feature affect the current Threat Model?**
N/A

aestene, 2024-05-08:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2024-12-03:
This is done

closed 2024-12-03 by tsundvoll