equinor/isar #612: ISAR should only focus on Tasks and Steps should be delegated to the translational layers

oysand, 2024-09-17:
**Describe the improvement you would like to see**
ISAR should only need to focus on Tasks and does not need to look at Steps. This way a lot of the functionality currently handled as steps should be handled as task.

**How will this change existing functionality?**

1. Differentiating types such as MotionStep and InspectionStep should be assigned to tasks instead.
2. The robot interface should be updated so that status is only requested for tasks. Meaning that mission_status and step_status can be combined to one status call that gives the status for all the tasks.

**How will this improvement affect the current Threat Model?**
N/A

tsundvoll, 2024-09-23:
To be discussed if the time to do this is now, or later. Depends on the ability to test it now.

tsundvoll, 2024-10-21:
ISAR steps ™  is removed https://github.com/equinor/isar/pull/617

closed 2024-10-21 by tsundvoll