equinor/isar #665: Introduce Unlocalised state

andchiind, 2024-12-06:
**Describe the new feature you would like to see**
The name of the state is not important, but we need to state that informs listeners on MQTT that the robot is not ready to run missions since it is no longer localised. It should not be able to leave this state until localisation is regained, which can either happen internally in the ISAR state, or via a new localisation endpoint if it is not able to handle it automatically.

**Describe the solution you'd like**
Based on how the robot responds when running a mission, ISAR implementations should be able to give feedback to the statemachine that it should move to a delocalised state, and this should then be reported as a robot state in the MQTT status publisher.

**How will this feature affect the current Threat Model?**
N/A

Eddasol, 2024-12-12:
Does it have to be unlocalized state? Can it be a more generic error state combined with a mqtt error message that gets propagated to the user?

andchiind, 2024-12-18:
> Does it have to be unlocalized state? Can it be a more generic error state combined with a mqtt error message that gets propagated to the user?

Yeah the name is not important, but we need a state to tell the user that the robot is not offline, not running missions, but also not able to run missions. So this could also for instance include that the P-stop is active, but we might want a discussion on how fine-grained we want the states/statuses.

andchiind, 2025-08-01:
This state has been surpassed by the InterventionNeeded state https://github.com/equinor/isar/pull/870

closed 2025-08-01 by andchiind