equinor/isar #685: Generate ISAR Task ID in ISAR

tsundvoll, 2024-12-17:
**Describe the improvement you would like to see**
Generate ISAR Task ID in ISAR. Currently this is accepted on the start-mission request. I want it removed from the StartMissionTaskDefinition and generated at Task instansiation instead.

**How will this change existing functionality?**
External applications (Flotilla) will get all the IDs they need from ISAR

**How will this improvement affect the current Threat Model?**
NA

tsundvoll, 2024-12-17:
This is dependent on Flotilla being ready to accept ISAR Task ID in the response from the start-mission endpoint

Eddasol, 2024-12-17:
Related issue: [#1888](https://github.com/equinor/flotilla/issues/1888)

Eddasol, 2025-06-18:
Suggest accept that isar uses the provided id and only generate new it none is received. This is the current implementation and the issue should therefor be considered closed as not planned

closed 2025-06-25 by tsundvoll