equinor/isar #772: Gets stuck in stopping state if new mission when returning home

Eddasol, 2025-04-30:
**Describe the bug**
Isar can get stuck in stopping, causing the robot to be stuck in ongoing from flotillas point of view.

**To Reproduce**
Steps to reproduce the behavior:
* Run mission
* Wait for return to home to start
* Schedule mission

This error only occurs if the new mission is scheduled right after the return to home started

**Expected behavior**
Shouldn't get stuck in stopping state. The scheduled mission should start

**Screenshots**
Logs
[image]

Looks like this before the queue times out
[image]

And looks like this after time out
[image]

Eddasol, 2025-04-30:
Update: Being quick is not critical for triggering the bug. Always occurs the second time a mission is scheduled during a return to home

oysand, 2025-04-30:
fixed in #774

closed 2025-04-30 by oysand