equinor/isar #777: Missions get stuck in the queue

mrica-equinor, 2025-05-12:
**Describe the bug**
When scheduling more than one mission, the second one seems to get stuck in the queue and the robot return home. The mission just lingers until something triggers on robot available again. 

Assume this is an issue between the even based communication between isar and flotilla.

**To Reproduce**
Go to dev and schedule two missions.

**Expected behavior**
Isar should notify flotilla thtat the rebot is available and the next mission in the queue should start. 

**Screenshots**
After the first mission finishes:
[image]

After the return to home finishes:
[image]

tsundvoll, 2025-05-12:
Consider checking with and without battery limits?

mrica-equinor, 2025-05-14:
Handled in PR https://github.com/equinor/flotilla/pull/2193