equinor/isar #808: Cannot know if robot is both home and protective stop

Eddasol, 2025-06-19:
**Describe the bug**
Currently both home and protective_stop are potential options for robot_status, but what if robot is both home and protective_stop is active. Should it say home or protective stop? This is up to the robot package but should have information of both these things

**To Reproduce**

**Expected behavior**

**Screenshots**

tsundvoll, 2025-06-25:
One approach can be to let the robot_status only be either normal, protective_stop or offline.

The home state could be entered when the robot successfully completes a return to home mission.

andchiind, 2025-09-08:
We an arguably move to intervention_needed state when we detect the P-stop. That way we don't need intervention needed state, and we will be able to make sure that the robot is docked before missions resume.