equinor/isar #819: Poll robot.robot_status() directly instead of polling shared state that updates every 5sec

Eddasol, 2025-06-27:
**Describe the improvement you would like to see**
Currently the states are reading shared_state.  Every 5 sec robot.robot_status gets read and is used to update shared_state. Suggest running robot.robot_status whenever the states ask for it instead so they dont use outdated value.

This should be implemented like task_status. The states will create a robot_status_request and that triggers a function that responds with new robot_status.

**How will this change existing functionality?**
Less chance of reading old values

**How will this improvement affect the current Threat Model?**

andchiind, 2025-08-01:
We could also avoid polling by checking the robot status when we enter a state and then only waiting for robot status update events from the robot service which would be sent only when the status changes. This would make the system more event based and remove a lot of excess checks.

tsundvoll, 2025-08-11:
> We could also avoid polling by checking the robot status when we enter a state and then only waiting for robot status update events from the robot service which would be sent only when the status changes. This would make the system more event based and remove a lot of excess checks.

I like this. Then we can name the events specifically after what is happening, instead of the generic robot_status_changed event, that does not tell us much on the diagram. I.e. "task_completed" or "mission_failed", instead of "robot_status_changed".