equinor/isar #883: Might send next mission before previous mission is completed

Eddasol, 2025-08-12:
**Describe the bug**
For some robots it takes time from completion of the last task until the mission completes (eg data upload). Since isar assumes the mission to be completed when all tasks are completed, it often sends the next mission while the robot still has a mission. This causes conflict and might cause the next mission to fail initiating.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
Next mission shouldn't be sent until previous mission is completed

**Screenshots**

Eddasol, 2025-08-12:
Currently handled by workaround in robot package. Workaround should be removed when issue is fixed on isar side. Workaround PR: https://github.com/equinor/isar-taurob/pull/625

andchiind, 2025-11-12:
Solved in recent refactor

closed 2025-11-12 by andchiind