equinor/isar #919: Return to home fails to stop

Eddasol, 2025-09-18:
**Describe the bug**
Return to home fails to stop if scheduling new mission while return home is getting scheduled.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**

**Screenshots**
Screenshots of the logs when this occurred on Taurob. An inspection mission was scheduled right as the return to home was starting/getting scheduled. The new mission just ended up in the queue. While the return to home was running a new mission was added to the queue in an attempt to stop the ongoing return to home mission. This also ended up in the queue. Had to wait until the robot were successfully docked before being able to get the inspection mission to run.

[image]

[image]

[image]

[image]

andchiind, 2025-11-12:
One issue is what the vendor API might do when receiving a stop request while starting a mission. I suggest having an "initialise_mission" state for when the mission is not ready yet. In this state we do not handle stop requests, and instead wait until the mission has started. This solves the issue for returning home cancellation when receiving a new mission (the robot will appear busy while starting the mission), and the user will get better feedback over why their stop request might fail, in the event that it times out since the mission takes too long to start. We could also consider buffering the stop request so that it will stop as soon as the mission has started.

andchiind, 2025-11-17:
In short: in robot.py, wait for the robot_start_mission thread to complete before sending the stop request to the robot. Adding a new state as mentioned in the previous comment is not needed for this issue.